Michael-F-Bryan/gcode-rs #2: Futures and multitasking

Michael-F-Bryan, 2017-05-24:
How do we deal with the fact that we could be reading data from a serial port (and therefore waiting for it to arrive) and may not have everything in memory? Should we block until we have the data, or should we use something like `futures` to allow multitasking?

etrombly, 2017-06-02:
I'm going to be using this with rtfm which will handle the multitasking. I was planning on having the serial read in line by line, and sending a whole line at a time to the parser. Not sure if that method will work for everybody though.

Michael-F-Bryan, 2017-06-02:
I imagine most embedded Rust programs would follow a similar pattern. 

Can you give me an example of how (in an ideal world) you'd like the API to behave? I was thinking you might create some "SerialReader" which yields characters and has an underlying buffer, then every time the buffer is empty it'll ask for more bytes from whoever is on the other end of the serial connection. But that may be entirely different to how you'd like to use the library...

etrombly, 2017-06-02:
I just got serial working on my board, so need to do some testing for the best way to implement it. I was thinking of making it RepRap compatible. So I'd read a line at a time, send that to process (just send x and y to the stepper driver in my case, for a more complicated firmware would need to go to the path planner), then send an ok response. Will have to see how much buffering is required and if sending an ok for each line adds too much latency. May eventually add SD support and just read directly from there. I don't have any experience in designing printer firmware, so I'm just learning as I go.

wose, 2017-06-02:
I didn't have time to think about the different modules (crates) which could be part of a 3D printer/CNC firmware yet. I hope I'll find some time this weekend. However, I'd like to suggest decoupling the whole serial and printer part from the gcode parser.

### In my perfect world...
...the printer firmware supports a simple serial protocol which encapsulates the gcode strings or some other data. It also allows communicating to the host if the printer is ready to receive new commands and acknowledges correctly received commands. It also allows to trigger resends if the checksum is wrong.

But all this should be independent from the gcode parser itself. It's just a serial protocol to transport the data. A useful protocol for this purpose probably already exists but may not be implemented in rust yet.

After the received gcode line is parsed into a proper command it is checked if the command is either an immediate command, like emergency stop, and processed or the command is something like a motion command and put into a command queue. This allows a motion planner to implement features like look-ahead. This will also allow the host to send new commands while the printer is processing the previously received commands until the queue is full. At this point it is not relevant if the gcode commands originated from a serial connection, from a file on a SD Card or some other source.

Then there is the printer part which will process commands in the queue which depends on the machine you're building (what hardware: stepper drivers, sensors, ...).

etrombly, 2017-06-02:
definitely, I wasn't saying you should add any of those features to this project. Just showing you how I intend to use it. My stepper controller will look similar to this, https://github.com/etrombly/bluepill/blob/master/examples/stepper_tasks.rs just replace the controller task with the serial input from here https://github.com/etrombly/bluepill/blob/master/examples/loopback.rs (still need to add a buffer). The gcode parser will be the step in between the serial and stepper tasks.

Michael-F-Bryan, 2017-06-02:
@wose, that sounds pretty much how I'd do things. I was thinking something like the `SerialReader` I mentioned earlier would act as "glue" to convert the data coming from *some* serial protocol into the `char` iterator the `gcode` crate expects.

It sounds like the printer firmware can be broken down into a couple fairly self-contained modules:

- Communication protocol
- Gcode parser
- Motion planner
- Motion "executor"/stepper controller

With some sort of meta-crate which contains all the trait definitions and glue necessary to bind everything together.

wose, 2017-06-02:
Oh, ok sorry, but it's awesome you two think the same way. I'm really looking forward to patch something like the above together.

Michael-F-Bryan, 2018-09-19:
The current implementation will parse consecutive gcodes out of an arbitrary `&str`, so should work just fine if reading line-by-line or in larger chunks. It doesn't really support reading from a stream-like source (à la `std::io::Read`), but when you're working with `no_std` that should probably be done at a higher level anyway.

closed 2018-09-19 by Michael-F-Bryan