Michael-F-Bryan/gcode-rs #5: M is both an argument and a command

Michael-F-Bryan, 2017-05-26:
There are several letters which can be both commands and arguments, depending on the context. It might be necessary to refactor the parser to take into account things like newline characters and what the last command was.

Here's an exerpt from an automatically generated program my roommate gave me.

```gcode
;FLAVOR:RepRap
;TIME:5373
;Filament used: 3.18051m
;Layer height: 0.1
;Generated with Cura_SteamEngine 2.4.0
M190 S90
M104 S235
M109 S235
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
```

But when you put it through the parser, the `M190` and other `M` commands get parsed as  arguments instead of commands.

```text
M190 S90 M104 S235 M109 S235    (line: 5, column: 1)
G21     (line: 8, column: 1)
G90 M82 M107    (line: 9, column: 1)
G28 X0 Y0       (line: 12, column: 1)
G28 Z0  (line: 13, column: 1)
G1 Z15 F9000    (line: 14, column: 1)
G92 E0  (line: 15, column: 1)
G1 F200 E3      (line: 16, column: 1)
G92 E0  (line: 17, column: 1)
G1 F9000 M117   (line: 18, column: 1)
```

Michael-F-Bryan, 2017-05-26:
Turns out an `M` code and a `G` code cannot be combined in a single instruction. So this should just be a case of removing `M` from the `arg_kind` rule in the grammar.

brainlag, 2017-05-27:
Isn't M not always a command? 
When you have code like this: 

G01 Y30 M8

then doesn't it basically mean:

M8
G01 Y30

Michael-F-Bryan, 2017-05-27:
Ah, originally I thought M could be an argument to a T command. 

E.g.:

```gcode
T01 M06
``` 
But turns out that's just preparing to change to tool 1 then calling M06 to actually swap tools. So it's two instructions put on the same line.

I think we used a machine-specific G code for changing tools when I did a unit on CNC and other machining processes, which may explain why I didn't pick this up.

closed 2017-05-29 by Michael-F-Bryan

benmkw, 2020-10-18:
I currently have a command like `M0 Turn ON 5000RPM` in gcode which I'm not sure how to handle correctly using this parser.

commit 4292a9a8 (2017-05-29): Removed M as an argument kind (fixes #5)