Michael-F-Bryan/gcode-rs #15: Different meanings for paramaters, depending on the G/M code

chickenchuck040, 2017-06-08:
I am working on a desktop gcode 3d viewer application using gcode-rs.

The issue I am running into is supporting all of the M codes and their parameters for 3d printers. I started to implement a couple of M codes on [my fork](https://github.com/chickenchuck040/gcode-rs/tree/mcodes). I have been using [the reprap wiki](http://reprap.org/wiki/G-code) as a refence, as the NIST gcode standard does not have most of the 3d printer mcodes.

I now have the problem that many of the M Codes use the same letter paramenter to represent different things.

For example, the S paramater could be a temperature, spindle speed, seconds, extruder number, retract length, etc

The issue becomes worse with paramaters like I, which go straight into a struct for G02 center. However, I could also be the I in PID, Inverse, or Idle factor.

I am thinking that we could either define each paramater meaning specific to each G/M code, or not assign any meaning at all.

Michael-F-Bryan, 2017-06-08:
I've been thinking for a while now that making a type-safe variant of each M or G code which exposes just their arguments is a little unrealistic, so I'm thinking of rearranging how the library works.

What if I were to expose G and M codes as something like this?

```rust
struct Command {
  kind: CommandKind,
  number: u32,
  args: Args,
  line_number: Option<u32>,
}

struct Args {
  x: Option<f32>,
  y: Option<f32>,  
  z: Option<f32>,
  ...
  s: Option<f32>,
}

enum CommandKind {
  G,
  M,
}
```

So in this way, the `Args` type just ends up as a statically sized bag of values which may or may not be present. This massively simplifies the parsing process for me, but you also lose a lot of the ability to ensure correctness at compile time, which is one of Rust's killer features...

closed 2017-06-09 by Michael-F-Bryan

chickenchuck040, 2017-06-09:
Representing Gcodes like that could definently work.

I have a couple of questions though:
The `number` field of `Command` is a `u32`. Are we handling gcodes with decimal places? For example `G38.2`.
Also, the `M117` Mcode has a single string argument. How are we handling that?

Michael-F-Bryan, 2017-06-09:
I just did a big refactor of the parser, among many other changes it now it exposes the `Command`'s `number` as either an integer or a decimal ([docs]). I need to update the lexer for it before it'll work correctly though.

I'm not sure about the best way to handle a string argument though. Because of the way I've done the parser and lexer, you could end up with ambiguities if it's also got to look out for arbitrary strings. In theory, if the lexer looks ahead and sees several consecutive letters I could treat everything until the next comment or newline as a string and emit that as a single token. 

It makes things interesting though because you now need to either have a dynamically sized string (not really possible for embedded stuff), or use a buffer (which then bloats everything else because of how enums work).

[docs]: https://michael-f-bryan.github.io/gcode-rs/gcode/parser/struct.Command.html

PR #16 (2017-06-09): Refactor
I've completely rewritten the parser layer for this crate and deprecated the old `low_level` module.

This fixes #15, also fixes #6.