Michael-F-Bryan/gcode-rs #17: Use bytes instead of char

Michael-F-Bryan, 2017-06-09:
At the moment, the only reason the lexer wants a stream of `char` is because I use a lot of simple string-y methods (e.g. to_upper and is_digit). There are probably some pretty decent speed improvements to be had if we were to accept a stream of bytes instead. It also makes things more convenient when you're working with files and serial connections.

luser, 2017-10-26:
FYI [`AsciiExt`](https://doc.rust-lang.org/std/ascii/trait.AsciiExt.html) should have everything you need to do this, but unfortunately some of the useful methods like `is_ascii_whitespace` [are currently unstable](https://github.com/rust-lang/rust/issues/39658).

Michael-F-Bryan, 2017-10-27:
It looks like `AsciiExt` is part of `std`, while this really needs to be a `#![no_std]` library so it can be used on microcontrollers.

I've actually started the parsing part of this library to make it work on bytes directly, I just haven't had time to finish it off and land the PR.

closed 2018-09-19 by Michael-F-Bryan