Michael-F-Bryan/gcode-rs #32: WebAssembly Showcase Website

Michael-F-Bryan, 2018-11-18:
It'd be cool to have some sort of showcase website which will interpret any `gcode` program from the browser using web assembly. 

The `gcode` crate itself only deals with gcodes at the syntactic level, so as well as the normal web app side of things, we'd need to create our own virtual-machine which can interpret gcodes and update internal state appropriately.

I've never made a website using WASM before, so any help is most appreciated!

A rough sketch of what I had in mind:

[image]

Michael-F-Bryan, 2018-11-23:
I think the big thing we'll need for this is to write a simulator system which can be compiled to WASM, then some sort of shell using an existing JavaScript framework (e.g. React).

@Dylan-DPC have you ever done anything with WASM or React?

Dylan-DPC, 2018-11-23:
@Michael-F-Bryan nope. Though i have worked with vuejs earlier. I can manage the WASM stuff.

seanlinsley, 2018-12-08:
there's probably a lot of code that can be re-used here:
- https://webassembly.studio
- https://github.com/wasdk/WebAssemblyStudio

etrombly, 2019-11-07:
I've made a super minimal initial version. Only parses G0 and G1 now, and there's a lot more to do. You can take a look here https://github.com/etrombly/gcode-yew

etrombly, 2019-11-14:
So the site can now do G0,1,2,3,90, and 91, so it can handle the test cases I threw at it. I'm sure there are still some bugs though. I also added drag and zoom support. Not sure if I'll work much more on it, but would take input on themeing or functionality.

commit bddf611a (2019-11-15): Mentioned @etrombly's showcase website (fixes #32)
closed 2019-11-15 by Michael-F-Bryan