Michael-F-Bryan/gcode-rs #35: reduce memory usage?

etrombly, 2019-04-17:
Previous versions were working pretty well on a stm32f103, with 20Kb of sram. Using the current version I get segfaults trying to parse even one line of gcode (it works fine on a chip with more memory). I noticed the default settings for the number of arguments in a line of gcode and number of commands in a block were both 10. This takes up about 4Kb per block. Are those sizes arbitrary? or were they based on typical usage? Here's the code I used to test https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=9728a5e97e95fd9d27e287b176696e4a . If setting both ArrayVecs down to 5 is ok it drops ram usage to a little over 1Kb.

etrombly, 2019-04-18:
Thought about this a little more, number of commands would need to be at least 7 or 8 still, something like a "G2 X0 Y0 I0 J0 E1 F1 Z1" is about the most number of args I can think of. Maybe add a lowmem feature and set number of commands to 1? Then you'd just have to document if you use the lowmem feature you have to ensure one command per line of gcode.  

Also if you could remove the spans from the arguments and gcode it would help a ton. With 10 arguments, 5 commands, and no spans, it takes it down to 576 bytes per block. It doesn't look like the spans are really integral to the parser.

Michael-F-Bryan, 2019-04-20:
> Using the current version I get segfaults trying to parse even one line of gcode (it works fine on a chip with more memory).

Just double-checking, but are the segfaults due to the `gcode` crate using more memory than exists, or could it be due to something else?

> I noticed the default settings for the number of arguments in a line of gcode and number of commands in a block were both 10. This takes up about 4Kb per block. Are those sizes arbitrary?

Yep, I think it'd be safe to lower some of the magic numbers used for buffer sizes. Cargo flags are meant to be composable, so I'd probably lower the magic numbers and make an (on by default) feature which increases the number of arguments/commands.

> Also if you could remove the spans from the arguments and gcode it would help a ton ... It doesn't look like the spans are really integral to the parser.

What about hiding span information behind an on by default `spans` feature? While they're not 100% necessary for a parser, having span information is super useful when you want to give feedback to the user (e.g. error reporting or highlighting the currently executing line in a simulator).

etrombly, 2019-04-20:
Yes, the segfaults are from running out of memory. If I set number of commands per block to 5 it runs fine. I think having these behind feature flags would work well. Not sure if you'd rather separate the options, or just have one lowmem flag.

PR #36 (2019-04-21): WIP: Memory usage
Refactor the crate to reduce the resource usage for memory-constrained devices.

Fixes #35.
closed 2019-04-21 by Michael-F-Bryan