Michael-F-Bryan/gcode-rs #43: Is there an example how to pass a gcode as an argument now that they are generic over arguments?

etrombly, 2020-03-04:
I tried updating to the most current version (master as of today). Gcodes have changed to be generic so I need to specify a type. I threw a random one in and got:
found reference `&gcode::gcode::GCode<arrayvec::ArrayVec<[gcode::words::Word; 5]>>

So I try setting it to that type and get:
   --> src/app.rs:233:58
    |
233 |     fn parse_G0(&mut self, code: &GCode<ArrayVec<[gcode::words::Word; 5]>>, context: &CanvasRenderingContext2d) {
    |                                                          ^^^^^ this module is private
    |

Any advice on how to handle it?

Michael-F-Bryan, 2020-03-05:
The `Word` type is [exported at the crate root](https://michael-f-bryan.github.io/gcode-rs/gcode/struct.Word.html).

I'm not sure what the best way to pass `GCode`s around is. In theory you *could* make your code generic over the underlying buffer type. In practice I think the best way is to use a default type parameter so you can refer to it as `GCode` instead of `GCode<ArrayVec<[Word.; 5]>>`. I should also add a recommendation to the `buffers` module docs for people to create typedefs for the various types if they need to override buffer sizes.

What are your thoughts?

etrombly, 2020-03-05:
So adding a "use gcode::Word" and changing it to Code<ArrayVec<[Word.; 5]>> worked. Maybe just having an example would be good enough? I don't really need it to be generic, since it's going to have the same type. Having a default type would be nice.

Michael-F-Bryan, 2020-03-07:
@etrombly, can you look at the API docs for the `gcode::buffers` module on docs.rs and tell me if they're good enough?

etrombly, 2020-03-09:
That looks good to me.

closed 2020-03-09 by Michael-F-Bryan