Michael-F-Bryan/gcode-rs #46: parser variable support?

strangesast, 2020-04-07:
I apologize if this is documented somewhere but I was hoping to parse a sample of gcode with variables.
```
#103=50.0 (BAR STOCK DIA);
...
G00 X[#103+3] Z5.0 M08;
```

Is there a way to handle these lines?  In my limited testing the parser ignores the declaration and raises a ParseFloatError on retrieval.  I'm not targeting an embedded system so I suppose I could handle the error then parse the span with regex or something.  Any input is appreciated.  Thanks!

Michael-F-Bryan, 2020-04-08:
Is this valid gcode? From memory, everything in gcode is hard coded and there are none of the higher level concepts like variables, loops, and conditionals that you find in real programming languages.

strangesast, 2020-04-08:
Yeah unfortunately (as I understand it) there are a bunch of variations in syntax from different manufacturers.  But I believe it's quite common to use values like tool offset, vice offset, etc in the gcode itself.  If CAM software (like fusion 360 or mastercam) is used, I believe you enter those values and compute the final value beforehand.

My use case involves a large library of gcode that was written by hand where parameters and [expressions](http://linuxcnc.org/docs/html/gcode/overview.html#gcode:expressions
) are common.

Interestingly (or horrifyingly), loops and conditionals are available in many dialects of gcode with WHILE, GOTO, and IF primitives available.

Anyway, thanks for the response.  I think I can parse most of it without much effort then use regex for the rest.

Michael-F-Bryan, 2020-04-09:
I intended the parser to be fault tolerant and skip over text it can't understand, so let me know if you encounter issues where some commands are broken after skipping variables or expressions (e.g. the previous line contained an expression and then we only parse half of the next line).

Depending on how much effort you want to go to, the [`Callbacks::unknown_content()`](https://docs.rs/gcode/0.6.1/gcode/trait.Callbacks.html#method.unknown_content) callback may be useful for detecting when an expression/variable was skipped and doing something to handle it. That callback gives you the "unknown" text and where it occurred, so in theory you should have enough information to figure out the rest.

closed 2020-04-09 by Michael-F-Bryan

strangesast, 2020-04-09:
Thanks for pointing that out.  That's what I'll do.  Thanks