Michael-F-Bryan/gcode-rs #50: Fusion generated gcode

benmkw, 2020-10-18:
the following input
```gcode
;Fusion 360 CAM 2.0.9011
; Posts processor: MPCNC ohne Tastplatte.cps
; Gcode generated: Sun Oct 18 21:46:19 2020 GMT
; Document: 5â€œ Quad frä¤sen v13
; Setup: Plates
;
; Ranges table:
; X: Min=1.341 Max=222.909 Size=221.568
; Y: Min=0.888 Max=227.927 Size=227.039
; Z: Min=-2 Max=15 Size=17
;
; Tools table:
; T3 D=2 CR=0 - ZMIN=-2 - flat end mill

; *** START begin ***
G90
G21
M84 S0
G92 X0 Y0 Z0
; *** START end ***

; *** SECTION begin ***
;Bore2 3 - Milling - Tool: 3 -  flat end mill
; X Min: 5.183 - X Max: 232.917
; Y Min: 10.385 - Y Max: 267
; Z Min: -2 - Z Max: 15
; COMMAND_START_SPINDLE
; COMMAND_SPINDLE_CLOCKWISE
M0 Turn ON 5000RPM
; COMMAND_COOLANT_ON
M117  Bore2 (3)
G0 Z15 F1200
G0 X230.263 Y14.512 F6000
G0 Z1.5 F1200
; MOVEMENT_LEAD_IN
G1 Z-0.3 F600
G1 X230.267 Z-0.339
G1 X230.278 Z-0.377
G1 X230.297 Z-0.411
G1 X230.322 Z-0.441
G1 X230.352 Z-0.466
G1 X230.386 Z-0.485
G1 X230.424 Z-0.496
G1 X230.463 Z-0.5
G1 X230.563
G2 X230.763 Y14.312 J-0.2
; MOVEMENT_CUTTING
G1 X230.759 Y14.249 Z-0.505
G1 X230.747 Y14.187 Z-0.51
G1 X230.727 Y14.127 Z-0.515
G1 X230.7 Y14.07 Z-0.52
G1 X230.666 Y14.016 Z-0.525
G1 X230.626 Y13.968 Z-0.53
G1 X230.579 Y13.925 Z-0.535
G1 X230.528 Y13.888 Z-0.54
G1 X230.472 Y13.858 Z-0.545
```

using
```rust
    for elem in gcode::parse(&data) {
        println!("{}", elem.to_string());
    }
```

produces 

```
G90
G21
M84 S0
G92
G92 X0 Y0 Z0
M0
T5000
T5000 R117 B2
G0 Z15 F1200
G0
G0 X230.263 Y14.512 F6000
G0
G0 Z1.5 F1200
G1 Z-0.3 F600
G1
G1 X230.267 Z-0.339
G1
G1 X230.278 Z-0.377
G1
G1 X230.297 Z-0.411
G1
G1 X230.322 Z-0.441
G1
G1 X230.352 Z-0.466
G1
G1 X230.386 Z-0.485
G1
G1 X230.424 Z-0.496
G1
G1 X230.463 Z-0.5
G1
G1 X230.563
G2
G2 X230.763 Y14.312 J-0.2
G1 X230.759 Y14.249 Z-0.505
G1
G1 X230.747 Y14.187 Z-0.51
G1
G1 X230.727 Y14.127 Z-0.515
G1
G1 X230.7 Y14.07 Z-0.52
G1
G1 X230.666 Y14.016 Z-0.525
G1
G1 X230.626 Y13.968 Z-0.53
G1
G1 X230.579 Y13.925 Z-0.535
G1
G1 X230.528 Y13.888 Z-0.54
G1
```

Which has issues around handling of M0 and double entries of G Lines.

It seems to me that Fusion outputs one gcode per line which this parser does not really take advantage of cause I think it assumes many instructions may appear on the line.

The double entries are probably some user error on my side :/ 

Thanks :)

Michael-F-Bryan, 2020-10-20:
I'm guessing you are running into #44. 

Something tells me I'll need to rewrite the parser when I get some free time. Or at least revise the grammar so newlines have a semantic meaning. That'll also give me a chance to handle your `M0 Turn ON 5000RPM` example in https://github.com/Michael-F-Bryan/gcode-rs/issues/5#issuecomment-711437435.

benmkw, 2020-10-20:
Ok nice 👍 

I can split by newlines and then parse each line which seems to fix the G ops appearing twice and also make them not get mangled by the M ones. (I'd like to plot them using wgpu+lyon much like the yew example and also display some statistics and possibly give the user some hooks for processing.)

FWIW this whole thing started for me cause a friend has trouble using another paid software with this gcode so I said well maybe I can help with that :D 

Feel free to ping me if you'd like me to test something or need feedback :)